lockevod/Karoo-KDoubleType #4: Many data fields are missing

kintrupf, 2025-01-06:
I guess it's still a work in progress to add other data fields 😉

These are the fields I'm missing:

- Elevation
- Ascent
- Descent
- Temperature
- Rear tire pressure
- Front tire pressure
- Heart rate in % max
- Distance remain
- Time to destination
- ETA

lockevod, 2025-01-06:
Yes, this field aren't in current version. I suppose you want these field in this order (for prioritize)

kintrupf, 2025-01-08:
I would need "Elevation" and "Heart rate in % max" first, because they would go on my main page.

nik-taylor, 2025-01-19:
I'd vote for temperature to be higher priority.  I came here just to do temp and gradient side by side

lockevod, 2025-02-01:
I'll add some of these fields in next 2.1 release

Jonathan-alter, 2025-02-03:
VAM 30s would be great too, but, VAM doesn´t show anything except "0"

lockevod, 2025-02-16:
Hello, I've added a new version with new measures and fixes. I'll close this ticket, please open a new one if you find any I have to fix.

Thanks

closed 2025-02-16 by lockevod