lockevod/Karoo-KDoubleType #6: Add a forth vertical or horizontal field

kintrupf, 2025-01-06:
If I use a 2 x 2 data field layout below or above the map (4 data fields in total) it look great if there would be also be 4 different vertical or horizontal double fields available.

lockevod, 2025-01-06:
Ok, I can do it. I will do it in v1.8 also

kintrupf, 2025-01-28:
Actually, I'd like to have a lot more vertical fields than four 😉
Four vertical fields would go to my main page with the map, but on a second page I could use them two, but of course with different contents. The aim here is to merge two pages with data fields by combining their data in double fields.

lockevod, 2025-02-01:
I've released a new version, please check the new one. You can select five vertical or horizontal fields and two rolling fields

closed 2025-02-01 by lockevod