lockevod/Karoo-KDoubleType #8: Issues running on K2

Matsalisbury, 2025-01-12:
Hi there,

Thanks so much for writing this.  I’ve had issues with this on the k2.  I’ve added 2 horizontal split fields on my map page
Ave speed, distance, ave power, ave speed. Only power shows correctly - others seem to be random numbers I can’t work out.  It also leads to crashes when loading the ride app

any debug I can do or tips?

cheers

Mat

lockevod, 2025-02-01:
Hello, I've released a new version, but I cannot check in karoo2, if you can provide me info about it

lockevod, 2025-02-16:
Hello, I've added a new version with new measures and fixes. I'll close this ticket, please open a new one if you find any I have to fix.

Karoo 2 has low resources and Extensions API has several perfomance issues, I'll try to improve kdouble, but i don't know if works fine in karoo 2(sorry I don't have one)

Please, check with 1-2 custom fields and check if it's working fine.

Thanks

closed 2025-02-16 by lockevod