lockevod/Karoo-KDoubleType #10: KDouble alignment

ajwalk3r, 2025-01-23:
Hi there awesome work with KDouble extension!
Just wanted to let you know that on the new Karoo 
For the horizontal fields alignment for me at least I can not select left or center it’s stuck on right and I can not turn on or off the use default Karoo alignment am I doing something wrong or is this a bug have a good day

lockevod, 2025-02-01:
Please could you check in the new release? Thanks

ajwalk3r, 2025-02-02:
It works kinda you did fix being able to choose center alignment as well as use Karoo default alignment but now every time you go into the ride And back out the alignment are different alignment so if you set it to center it takes a few attempts of going into the ride and backing out to get the alignments to match especially if you have multiple kdouble  fields

lockevod, 2025-02-03:
I think I've found where is the problem. I'll try to solve in v2.0.2 (this week)

lockevod, 2025-02-16:
Hello, I've added a new version with new measures and fixes. I'll close this ticket, please open a new one if you find any I have to fix.

Thanks

closed 2025-02-16 by lockevod