lockevod/Karoo-KDoubleType #15: Some small issues with the 2.0.1 release

kintrupf, 2025-02-01:
Just a few small issues with the last release, nothing major:

- Field "Distance remain" does not work, always shows 0. It's the second field on the right from the top. Should read around 45 at this point of my route.
- Horizontal aligment of the KDouble field at the right bottom does not work. Should be centered like the other fields.
- Why are the top corners of the fields with the background color rounded? Karoo limitation?

[image]

lockevod, 2025-02-01:
Hello, first and second solved ;) Second I've changed something, but please.. because I've checked before and i don't have this issues

Third... karoo has rounded corners.. but i don't understand why you don't have in top fields. I'll close this ticket but I'll try to check if I can check/uncheck this (for exemple, karoo 2 doens't have round corners)

closed 2025-02-01 by lockevod

kintrupf, 2025-02-01:
Thanks for your quick response!!! I just installed version 2.0.2, but unfortunately, the alignment issue is still not fixed for me 🥺
However, it only affects KDouble Field 1. If I exchange KDouble Field 1 with KDouble Field 4 it works.

The field on the bottom right is KDouble Field 1, should be centered like the other two:

[image]

lockevod, 2025-02-01:
Thanks, i'll review on monday... I'll fix other issues Kdouble has ... If you view something else, please report it ;)