lockevod/Karoo-KDoubleType #17: Incorrect Field Assignment in Settings

p18dh28ejfi, 2025-02-12:
There are issues with the correct assignment of settings in the extension:

1. Field 3 and Field 4: The selected settings are not applied correctly. For example, when setting **Grade** and **Headwind**, the display instead shows **Headwind** and **TSS**.
2. "Horizontal Fields": This setting is not being applied on the fields 3+4.
3. Field 4 & Field 5 Dependency: Changing the Field 5 setting unexpectedly modifies the Horizontal Field setting of Field 4.

Expected Behaviour

- The selected settings should be applied exactly as chosen in the menu.
- Changing one field’s setting should not affect other fields unless intended.

Steps to Reproduce

- Open the extension settings on the Hammerhead Karoo 3.
- Assign specific values to Field 3 and Field 4 (e.g., Grade and Headwind).
- Save the settings and check the display.
- Modify the "Horizontal Fields" setting and observe if the changes are applied.
- Change Field 5 and check whether Field 4 is unexpectedly affected.

Screenshots
[imgur](https://imgur.com/a/boocLxe)

System Information
Device: Hammerhead Karoo 3
Firmware Version: 1.538.2049.20
Extension Version: 2.0.2

Additional Notes
This issue appears to be related to incorrect field mapping, possibly due to faulty storage or processing of the configuration.

lockevod, 2025-02-13:
Thanks for your reporting. I have a new version (next week release 2.1.0) but I haven't check this. I'll check general features an I'll correct this behaviour in 2.1.1

lockevod, 2025-02-16:
Hello, I've added a new version with new measures and fixes. I'll close this ticket, please open a new one if you find any I have to fix.

Thanks

closed 2025-02-16 by lockevod