lockevod/Karoo-KDoubleType #20: Headwind not appearing in 2.1

nik-taylor, 2025-02-16:
Hi,

Just installed 2.1 and even though I have headwind toggled in settings it doesn't appear to be a selectable field.

If possible, could you also add temperature as a field.

Thanks

lockevod, 2025-02-16:
Thanks.. it appears .. but i have an issue with the text.. sorryy... last text isn't for headwind.. if for divider (i'll correct now), please select the correct one, and press save 

I'll include temp in 2.2 version ( text correction will be 2.1.1)

nik-taylor, 2025-02-16:
Thanks for the fast turnaround

p18dh28ejfi, 2025-02-16:
I have 2.1.1 installed and can't find it in the fields-tab dropdown menu (no matter if I switched the slider in the conf.-tab or use horzontal fields (on/off)). 

Headwind shows up in the rolling-tab fields.

lockevod, 2025-02-22:
I think it's solve in 2.2.0 release

closed 2025-02-22 by lockevod

nik-taylor, 2025-02-23:
Seems to be working now.  Thanks.