lockevod/Karoo-KDoubleType #26: Double field with Speed and Distance doesn't display data correctly

formula155, 2025-03-14:
I have A Field set up as follows:
First Field - Speed, no colored zone
Second Field - Distance, no colored zone

Horizontal field is off

When this field displays on a map page, the Speed field displays 0.3, and never changes, and the distance field only displays the integer part of the distance.

Hope this is the right way to provide feedback, as you requested.

formula155, 2025-03-17:
I am using a Karoo 3, running ng latest version 1.544.2081. I'll give the new release a try,

Thanks,

Guy

formula155, 2025-03-20:
Hi,

I tried the new release, and issues still persist, although I did notice at one point that the Speed was varying from 0.3 to 0.4 and 0.5,

Thanks,

Guy

lockevod, 2025-03-21:
Hello, I have the same configuration and it's working fine, could you give more information. For example, do you have metric or imperial ? Could you use a normal speed field and check the "good" speed and give this value? And if you can upload some image.

formula155, 2025-03-22:
Hi,

I will not be able to ride for a few days to confirm the normal speed field, but I believe it is correct - I have a profile that uses the standard field and last time I looked it was doing the right thing.

And regarding Imperial vs metric, I am using Imperial (mph),

Thanks,

Guy

lockevod, 2025-03-22:
I suppose the issue is with conversion between imperial and metric. I’ll try to recheck

Regards

lockevod, 2025-03-23:
Hello, i'm sure the "orginal speed" in karoo is working fine.. my question is if you can say me when you have 0,3-0,4 what is your real speed ? I don't have imperial units and I suppose this is the problem (because karoo documentation is a little...)

Regards

formula155, 2025-04-15:
Did finally get to ride, and data is mow displayed correctly. Distance is shown as an integer, i.e no digits after a decimal point[int - is this expected?