lockevod/Karoo-KDoubleType #33: Bug: Incorrect Zone Color Display at Zone Boundaries

jpweytjens, 2025-04-09:
## Bug: Incorrect Zone Color Display at Zone Boundaries

### Device and Version Information
- Device: Hammerhead Karoo 3
- Extension Version: 3.1.0 (Latest)

### Issue Description
When using the KDouble Type extension, colored fields incorrectly display dark red (the color for a zone 7 high intensity) when values are at or near the boundary values of lower zones. This creates a confusing user experience as metrics appear to be in a high-intensity zone when they are actually at the boundary of lower intensity zones.

### Problem Location
The issue is in the zone calculation logic in `com.enderthor.kCustomField.extensions.Zones.kt`. Specifically, the `getZone` function has several problems in how it determines which zone a value belongs to:

1. It unnecessarily floors the value before comparison (`val valueabs = floor(value.absoluteValue)`)
2. It has problematic boundary condition handling with `valueabs >= min && valueabs < max` which excludes exact boundary values from the lower zone
3. It defaults to Zone7 (dark red) when it can't find a specific zone: `return zoneList.getOrNull(i) ?: Zone.Zone7`

### Proposed Fix
The problem can be fixed by:

1. Removing the unnecessary flooring of values, since we're comparing doubles to doubles
2. Improving the boundary condition check to properly handle exact matches
3. Using more appropriate default zones instead of defaulting to Zone7 (high intensity)

Here's a suggested fix:

```kotlin
inline fun <reified T> getZone(userZones: List<T>, value: Double): Zone? {
    val zoneList = zones[userZones.size] ?: return null
    val valueabs = value.absoluteValue  // Remove flooring to preserve precision
    
    // Check if value is in any defined zone
    for (i in userZones.indices) {
        val zone = userZones[i]
        val min = when (zone) {
            is UserProfile.Zone -> zone.min.toDouble()
            is zoneslope -> zone.min
            else -> return null
        }
        val max = when (zone) {
            is UserProfile.Zone -> zone.max.toDouble()
            is zoneslope -> zone.max
            else -> return null
        }
        
        // Simplified boundary condition - include the max value in the current zone
        if (valueabs >= min && valueabs <= max) {
            return zoneList.getOrNull(i) ?: Zone.Zone1  // Default to Zone1 (less intense) instead of Zone7
        }
    }
    
    // Handle values beyond the defined zones
    val lastZone = userZones.lastOrNull() ?: return null
    val lastMax = when (lastZone) {
        is UserProfile.Zone -> lastZone.max.toDouble()
        is zoneslope -> 99.0
        else -> return null
    }
    
    if (valueabs >= lastMax) {
        return zoneList.lastOrNull() ?: Zone.Zone8
    }
    
    // If no zone matched, return a moderate zone rather than highest intensity
    return Zone.Zone3
}
```

This fix should provide more accurate and consistent zone coloring at boundary values and prevent the incorrect display of dark red (Zone7) for values that should be in lower zones.

### Expected Behavior After Fix
After applying this fix, values at zone boundaries should correctly show the appropriate zone color rather than incorrectly defaulting to the high-intensity red color. This will provide a more accurate visual representation of intensity zones during rides.

lockevod, 2025-04-09:
Thanks for your proposal. I'll apply in next release ;)

jpweytjens, 2025-05-06:
Hey @lockevod , when do you plan to publish a new release?

lockevod, 2025-05-22:
I think this week or next week, I'm adding a buttom to simulate (try to .. because karoo has a buzzer only) a bell sound ...